sumotoy/TFT_ILI9163C #1: BLACK Tag display issue, not working properly

sumotoy, 2015-02-25:
Can you provide a datasheet for this display?

DDimitr, 2015-02-26:
Assuming that you are referring to a black PCB TFT sourced from Ebay, this is the documentation that was provided by the Ebay seller: https://www.dropbox.com/s/fwcoqy08thh5dzn/1.44Black_SPI.zip?dl=0

[image]
[image]
[image]
[image]

sp9mrn, 2015-02-26:
In my case "black" LCD has shifted display after first start. After tests -  BLACK does NOT need offset correction in library. 
I mean "offset" in "TFT_ILI9163C.h" shold be set to "0". Very good written library, clear and easy to make own corrections.

sumotoy, 2015-02-26:
Oh, thanks! You confirm it works with what processor?
I have seen a Black tag that has the surrounding white display colored as Black that need offset.
At this point I presume there's more strain around!
Btw, I'm gonna update the whole library in  these days since it's an early alpha with the latest stuff (SPI transaction, etc) and try to figure out why with very fast MCU the screen refresh (done at high speed) produce a visible flash (I can see with the 3d example and Teensy3), also I'm plannig add a serious character support.

MasudaNaika, 2015-02-26:
I forked your ILI9193C libray to use with my Arduino compatible boards. It uses AVR MSPIM.
https://github.com/MasudaNaika/TFT_ILI9163C
https://github.com/MasudaNaika/powerpack7

I tested 1.44 RED PCB (with wrong config) form ebay supplier and M014C9193SPI lcd module available in Japan.
http://www.aitendo.com/product/3857
M014C9193SPI module has correct configuration and the OFFSET should be zero to work.

I also noticed, even if with RED PCB, that const _GRAMSIZE = 128 * 128 is enough to fill screen with your offset workaround.

mbed (Nucleo F411RE) version is here
http://developer.mbed.org/users/peu605/code/TFT_ILI9163C/

Thank you summotoy.

sp9mrn, 2015-02-26:
[image]  LCD connected to arduino mini 3,3v/8MHz. Also no problems with nano3 and uno. (with level shifters). 
1. I never tried to work with 5V - afraid to fry
2. Didn't try witn pro micro or leonardo (yet)

castingflame, 2015-02-26:
I'll reload my setup and take a look at the offset sp9mrn mentions. Looking like a result!

sumotoy, 2015-02-26:
I just finish to setup a demo board (tomorrow I will post it) for update the library, if anyone have suggestions I will happy to include.
Till now here's my goals:
1) New SPI functions, bit more fast and SPI transaction compatible
2) Compatibility insured when other SPI devices uses same line (that's why I have to modify some SPI command)
3) Support for user character set.

castingflame, 2015-02-26:
Its all a bit over my head but im learning fast and appreciate all your work for the community.

MasudaNaika, 2015-02-28:
ILI9163 accepts continuous SPI data without toggling CS.

In ILI9163 library, most frequent called function are "setAddrWindow" and "fillRect".
If MCU's SPI has FIFO buffer, keeping CS low and sending continuous data  may improve performance.

e.g.
fillRect:  DC hi-> CS low -> send n of data -> CS high

An screen shot in below link may help to understand : First data has 4 bytes lengh with continuous 32 SCLKs. No pause in a transaction.
http://developer.mbed.org/media/uploads/peu605/ili9163_f411.jpg.pagespeed.ce.I2jsiCXX1g.jpg

In my case, 50MHz SPI clock worked properly on RED PCB.

Of course as you did, I think using Freescale's DSPI is the best solution.
Thanks again for your great work.

sumotoy, 2015-02-28:
The new SPI code for Teensy do that, but I will try to use SPI DMA in the next days.
Thanks for info, I'll add the posting in the readme

MasudaNaika, 2015-02-28:
Oops!, you have already implemented, excuse me...

sumotoy, 2015-02-28:
Shinpai shinade...

sumotoy, 2015-02-28:
ok, issue closed I think....

sumotoy, 2015-03-01:
Masuda, I've corrected some stuff that was wrong, please check code and in particular fillScreen so if you ported the code you should reflect changes.
About GRAMSIZE, you can leave at 128x128 if you do not plan to use scroll (that I've just added). If so, scroll will use the off-screen ram (in short all mapped ram) and you prolly have garbage during scroll in one side of the screen!

MasudaNaika, 2015-03-01:
Sumotoy, thank you inform me, I'lll check it.

MasudaNaika, 2015-03-02:
I found you commented out,
```
void TFT_ILI9163C::defineScrollArea(uint16_t a,uint16_t c){
	//0,2
		uint16_t b;
		if ((a+c) > _TFTWIDTH) return;
		b = (_TFTWIDTH - (a+c));
		SPI.beginTransaction(SPISettings(SPICLOCK, MSBFIRST, SPI_MODE0));
		writecommand_cont(0x33);
		if (rotation == 0 || rotation > 1){
			writedata16_cont(a);
			writedata16_cont(b);
			writedata16_last(c);
		} else {
			writedata16_cont(a + __OFFSET);
			writedata16_cont(b + __OFFSET);
			writedata16_last(c + __OFFSET);
		}
		endProc();
		setAddr(0x00,0x00,_GRAMWIDTH-1,_GRAMHEIGH-1);
		setRotation(0);
		
}
```

According to the datasheet,  
"The 3rd & 4th parameter VSA[15...0] describes the height of the Vertical Scrolling Area (in No. of lines of the Frame Memory [not the display] from the Vertical Scrolling Start Address)"

```
	writedata16_cont(a + __OFFSET);
	writedata16_cont(b);	// <-
	writedata16_last(c);	// <-
```
Though I could not understand well what datasheet says , I will try it tonight.

Sincerely

sumotoy, 2015-03-02:
This is very important!

void TFT_ILI9163C::clearScreen(uint16_t color) {
	int px;
	#if defined(__MK20DX128__) || defined(__MK20DX256__)
		SPI.beginTransaction(SPISettings(SPICLOCK, MSBFIRST, SPI_MODE0));
		writecommand_cont(CMD_RAMWR);//this was missed!
		for (px = 0;px < _GRAMSIZE; px++){
			writedata16_cont(color);
		}
		_setAddrWindow(0x00,0x00,_GRAMWIDTH,_GRAMHEIGH);//go home
		writecommand_last(CMD_NOP);
		endProc();
	#else
		writecommand(CMD_RAMWR);
		for (px = 0;px < _GRAMSIZE; px++){
			writedata16(color);
		}
		setAddr(0x00,0x00,_GRAMWIDTH,_GRAMHEIGH);//go home
	#endif
}

Now it's correct

MasudaNaika, 2015-03-02:
Sumotoy, I've just tried scrolling.

I keep GRAM_SIZE = 128 * 128 and issue CMD_VSCLLDEF and CMD_VSSTADRS commands with OFFSET.
http://developer.mbed.org/users/peu605/code/TFT_ILI9163C/rev/1ebc27775d92

I've updoaded video to youtube.
http://youtu.be/wrQks3Npu1U

With Nucleo's SPI, I have to wait SPI become free every time toggling CS/DC.
If it's alright, translate to your code.

Best Regards

MasudaNaika, 2015-03-02:
Umm, on my F411RE, without DMA is faster except fillScreen :-<

<Nucleo F411RE DMA interrupt>
Benchmark                Time (microseconds)
Screen fill              21032
Text                     11223
Lines                    34808
Horiz/Vert Lines         7809
Rectangles (outline)     7643
Rectangles (filled)      5896
Circles (filled)         19582
Circles (outline)        21612
Triangles (outline)      14020
Triangles (filled)       19307
Rounded rects (outline)  16900
Rounded rects (filled)   52690
Done!

<Nucleo F411RE SOFT>
Benchmark                Time (microseconds)
Screen fill              26290
Text                     6055
Lines                    28571
Horiz/Vert Lines         2609
Rectangles (outline)     2448
Rectangles (filled)      40189
Circles (filled)         14520
Circles (outline)        20814
Triangles (outline)      8331
Triangles (filled)       14443
Rounded rects (outline)  11330
Rounded rects (filled)   47627
Done!

sumotoy, 2015-03-03:
Have to get back about clearScreen, the datasheet says one thing but in reality it's not!
Actual code works just fine (0.6b2) but I'm try to figure out why I need 2 commands and 6 data for simply write in ram! I'm sure there's a better and faster way to do that...
I'm sure that flashing on the botton of screen when I drive tft fast have something to do with this.

sumotoy, 2015-03-03:
About SPI, in Teensy my tests showed that DMA it's overall faster but I have 50% slow down on circles (outline)

MasudaNaika, 2015-03-03:
Good morning, sumotoy.

About clearScreen, I believe GRAMSIZE = 128 * 128 is enough to fill visible area. Clearing invisible VRAM area consumes more time. 
If you use scroll, scroll area tfa should be tfa height + OFFSET, because a TFA value is the absolute address of frame memory.

I am speculating the actual configuration of RED PCB as below.
[image]

Any suggestions would be greatly appreciated.

sumotoy, 2015-03-03:
It's bottom aligned, another fail of some chinese engineer. Infact I just got other 2 different display today (one red pcb and one black from 2 different vendors) correctly top aligned as it should be.

MasudaNaika, 2015-03-03:
I draw the short straw...

sumotoy, 2015-03-03:
I've fixed finally clearScreen!
'''
void TFT_ILI9163C::clearScreen(uint16_t color) {
	int px;
	#if defined(__MK20DX128__) || defined(__MK20DX256__)
		SPI.beginTransaction(SPISettings(SPICLOCK, MSBFIRST, SPI_MODE0));
		writecommand_cont(CMD_RAMWR);//this was missed!
		setAddr(0x00,0x00,_GRAMWIDTH,_GRAMHEIGH);//go home
		for (px = 0;px < _GRAMSIZE; px++){
			writedata16_cont(color);
		}
		_setAddrWindow(0x00,0x00,_GRAMWIDTH,_GRAMHEIGH);//go home
		writecommand_last(CMD_NOP);
		endProc();
	#else
		writecommand(CMD_RAMWR);
		setAddr(0x00,0x00,_GRAMWIDTH,_GRAMHEIGH);//go home
		for (px = 0;px < _GRAMSIZE; px++){
			writedata16(color);
		}
	#endif
}
'''

MasudaNaika, 2015-03-04:
Thank you hard working, sumotoy.

In my case, if limiting scroll area using defineScrollArea(0, 0) ( = CMD_VSCLLDEF, tfa =OFFSET, vsa =128, bfa=0), I think any garbage won't appear. 
I added "defineScrollArea(0, 0)" to TFT_ILI9163C::chipInit as default.

Youtube video shows vertical rolling scroll without any garbage. (but as you said, absolute scrolling axis is same in any rotation setting...)

sumotoy, 2015-03-04:
Good catch! I think all issues are gone now, ready for the Gold Master!

closed 2015-03-04 by sumotoy

MasudaNaika, 2015-03-05:
Thank you sumotoy, this is a very nice library.
(If you don't mind, please try GRAM_SIZE = 128 * 128 again. I hope you cannot  see any undesirable data during scroll.)

sumotoy, 2015-03-05:
I tried but with bottom aligned display seems don't work, got garbage. Maybe the only workaround it's clear the ILI ram in the init GRAM_SIZE = 128 * 160 then back to GRAM_SIZE = 128 * 128.

MasudaNaika, 2015-03-05:
Oh, it is disappointing. OK, Thanks for making time for me.

sumotoy, 2015-03-05:
A pleasure Masuda, you help a lot!