sumotoy/TFT_ILI9163C #3: How to connect it?

tymek22, 2015-03-11:
Hi,
The pins of those displays have other names than the ones of an Arduino, so I'm a bit confused about the connections. Does anyone know how to connect the pins of the module to an arduino?

Thanks in advance.

sumotoy, 2015-03-12:
Wich arduino?

sp9mrn, 2015-03-12:
Specify problem - I've got sck, sda, a0, reset, cs - and of course led, vcc gnd - but that is clear.

sample (Arduino mini) (old library)
//*******************1,44 LCD****************************************
#define __CS 10
#define __DC 9
#define __RST 8
TFT_ILI9163C display = TFT_ILI9163C(__CS, __DC, __RST);
// piny 13 i 11 to sprzętowe SPI i nie trzeba ich definiować /hardware spi do not need to define

MAc

sumotoy, 2015-03-13:
MOSI,MISO and SCK are hardware PINS and in UNO:
SCLK:13
MOSI:11
MISO:12
In MEGA are different, DUE different too, etc. please refere to the board specification you have.
Also notice that if you have UNO and you use for CS a pin different from 10 you probably need to add in setup pinMode(10,OUTPUT); and digitalWrite(10,HIGH); even if you don't use it, it's a well know bug of the Arduino SPI library.
This not happen with Teensy 3.1 and DUE.

tymek22, 2015-03-13:
Ok thanks for your answers, but to which pins on the arduino should I connect sda and a0? And what is DC?

tymek22, 2015-03-13:
Ow sorry! I dind't see that you updated the readme file and now I see how to connect it. 
Thanks!

closed 2015-03-13 by tymek22