sumotoy/TFT_ILI9163C #8: Drawing line doesn't fill fully line space

Puchaczov, 2015-06-14:
On arduino uno, there is a problem with drawing line with specific lenght

for example such code: 
  tft.drawFastHLine(0,0,128, RED);
doesn't fill in 100% line but there is 1 or 2 pixels with aren't filled by this line, same with VLine. 

It works when trying to supplement this pixels by drawPixel(...)

sumotoy, 2015-06-14:
Did you try the new 1.0 preview?
https://github.com/sumotoy/TFT_ILI9163C/tree/Pre-Release-1.0r

Btw, I know where's the problem, I fix the old one

Puchaczov, 2015-06-14:
In preview version drawFastHLine works correctly but vertical is still wrong.

tft.drawFastVLine(63,0,128,RED); I think the last pixel isn't filled correctly.

sumotoy, 2015-06-14:
ok, thanks for digging this, I'll fix tonite.

sumotoy, 2015-06-14:
tft.drawFastVLine(tft.width()/2,0,tft.height(),WHITE);

1 pixel loose, right?

sumotoy, 2015-06-14:
It look fixed now. There was a problem similar with drawPixel as well.
I wait confirm...

Puchaczov, 2015-06-15:
yea, looks fixed for me too, Great job and thanks :)

closed 2015-06-15 by Puchaczov