sumotoy/TFT_ILI9163C #9: Rought estimate for Energia compatible version?

fmetzler, 2015-08-30:
Hi,

This looks like an excellent project. I'd love to use your library to hook up my Adafruit TFT shield with my Redbearlab CC3200 (using the Energia IDE).

Do you have a rough sense when an Energia compatible version will be ready? Unfortunately, I don't have the skills to port it myself. 

Thanks a lot!

sumotoy, 2015-08-30:
I don't have Redbearlab CC3200, have only the old Stellaris, btw I'm working on an importan update of this library so can you point me to a TFT library that works 'fast' with your CPU as study?

ortszentrum, 2015-08-31:
Thanks for your comment. I haven't managed to get any TFT to work with the RBL CC3200 so far. I tried to use the 1.8" Adafruit TFT with the corresponding library from Adafruit and yours which I understand is based on it but I ran into difficulties with SPI and PROGMEM access.

sumotoy, 2015-08-31:
Redbearlab CC3200 it's similar to Stellaris? (apart the wifi stuff obviously)

tanah, 2015-10-18:
I'm also interested in this port, I have a Tiva C series Launchpad (TM4C123G). I would gladly help to make it happen but I don't have the knowledge to develop it on my own, so some clues on where to get started or points to solve the compatibility issue would be really helpful.

sumotoy, 2015-10-19:
I don't own a Tiva C as well, the Energia support it's painful, many processors, many differences.
It can help if you can point me to a working library that uses SPI.

closed 2015-11-28 by sumotoy