sumotoy/TFT_ILI9163C #11: cp437 fix

sincetomorrow, 2015-10-17:
Hi!

Great library, it works great! 
Have you integrated the glcd font bug with the missing character?
https://github.com/adafruit/Adafruit-GFX-Library/issues/22

Thank you very much,
Panos

sumotoy, 2015-10-17:
Ah, ok. I'll fix this later.
In the next version of this library I will drop adafruit GFX, you will able to use your own font.

sincetomorrow, 2015-10-17:
Hi again! Thanks for your quick replay. I have just noticed that I am using you newest library, in https://github.com/sumotoy/TFT_ILI9163C/tree/Pre-Release-1.0r, I have posted in this by mistake! In your new prelease, how do I access the degree sign?

Thanks a lot.

sumotoy, 2015-10-17:
try xxx.setFont(GFXFONT_GLCD_ASCII);
The 1.0preview it's not the last code I'm working on, a brand new preview it's coming in some days and will use a different font scheme.
Le me know if still missing the char.

closed 2015-11-28 by sumotoy