sumotoy/TFT_ILI9163C #12: Unable to drive TFT and SD card at the same time

vmiceli, 2015-11-16:
Hello folks,

I'm using the TFT_ILI9163C library (downloaded from here a couple of days ago) and an Arduino nano.
I'm using the following:

for the Display
#define __DC 9
#define __CS 8 //CS for TFT display
// MOSI --> (SDA) --> D11
#define __RST 12
// SCLK --> (SCK) --> D13

and pin 7 as CS for the SD card board.

With these settings I can run the library demos for both devices independently, However if I use both at the same time, the SD card will continue to work but the display will just turn white. It appear to me that once the SD.Begin(7) is issued the TFT display goes white.

Is this a known behavior? I've been trying for a couple of days already but no joy. Changing the CS pins (declaring them accordingly) makes no difference.

Thanks!

Enzo

sumotoy, 2015-11-16:
witch SD library are U using?
This library uses SPI transaction and expect that the other SPI libraries use SPI transaction as well to protect SPI communication between devices
If you are using an old SD library I suggest to update or use SDFat (and enable internally SPI transactions).
Another good thing it's pullup both CS lines with a 10K resistor, this force both devices inactive during startup.

vmiceli, 2015-11-17:
Thanks Sumotoy, I was using SDFat but didn't enable SPI transactions, That didn't do the trick though. I also added the pull up resistors. What made it work was to tie the RESET pin to 3.3V as opposed to pin 12 as I think that conflicts with the MISO line. I obviously needed to change the SPI transaction too.

Thanks a million!!

Enzo

closed 2015-11-18 by vmiceli