sumotoy/TFT_ILI9163C #13: sd wav player & ili9163 not working together

ramonschepers, 2015-11-21:
i noticed in your version, the audio library (sd wave player example) does not work correctly.
once i start the sd, i cannot get the screen to do anything at all.
there need to be changes in the way it talks to the spi port.

(sorry, i'm too tired to do a pull request and commit changes right now ^^; )

sumotoy, 2015-11-21:
I'm coding new SPI procedures that maintain the advantage of the SPI transactions, the original ILI procedures where developed with speed as main features but now there's other libraries that uses the same workarounds so conflicts are inevitable, expecially when both uses direct access to registers.
In my last code there's a problem with a SPI buffer tha should be checked constantly but (for speed) I have ignored, this causes a break in the middle of SPI transaction and everithing stop.
The best approach is fix this (I'm not sure if SD has the same problem), I will post a new code as soon I've finishing my tests, in the maintime you can post your suggestion as commit.

sumotoy, 2015-12-06:
I have used succesfully SD+Audio board+ILI, just used the Paul version of the SD.
However not all the SD card I have working, my samsung one high speed refuse to work and an unbrandad one worked with no problem.
Bill Greyman gived me some advice about SD cards, some has SPI weaks so connecting several SPI devices in the same line can be a problem, the best way I found is testing several SD cards and testing with an oscilloscope the waveform, as I suppose some card showed distorted waveform!
Maybe internally they have protection resistors, I don't know. I have noticed (by looking inside) that some commercial product use a buffer chip...

closed 2015-12-06 by sumotoy