sumotoy/TFT_ILI9163C #17: esp8266 ili9163 stops working after AP connection

gjt211, 2015-11-29:
I have been testing your pre-release 1.0r library with an ESP8266 12 and it works exceptionally well.
However, when I connect to an AP, the library ceases to function. No commands work.
There are no additional items physically wired to the ESP module, except for two LED's.

Can you provide some advice?

sumotoy, 2015-11-29:
There's an a wifi scan example and it works, it uses some functionality of the wifi but I never try an AP connection, maybe the AP it's blocking so I may need to add some Yeld command to free up some resources for library. I will look into this now.

sumotoy, 2015-11-29:
Ok, maybe I got it. I'm actually using eagle_soc.h for GPIO but seems broken when ESP8266 using some low level libraries, this is an issue of the ESP development that it's in early stage but in the maintime I try to modify this library to use a compatible mode, in some hour I will post an update.

sumotoy, 2015-11-29:
Try last version (but check first settings file and try to enable _ESP8266_STANDARDMODE.

gjt211, 2015-11-29:
Wow, thanks so much for your extremely fast response.
I will try it out today and post a reply about what happens.
Thank you again.

gjt211, 2015-11-29:
Hi Max, Great!!
It seems to be working now with the _ESP8266_STANDARDMODE enabled.
I will do some more tests later and update if I find any further issues. So far I am only using text.
Excellent work,

sumotoy, 2015-11-30:
Ok, so have to find another compatible alternative to eagle_soc.h, sigh!

closed 2015-11-30 by sumotoy