sumotoy/TFT_ILI9163C #18: display stop working randomly

smetis, 2015-12-01:
Hi,
I am trying the code on ESP8266 12E on nodemcu board. 
I am able to run sample codes, but they always end up with some distorted display or white screen. Sometimes it happen right away, sometimes after a minute, sometimes after an hour. 

Is there any known issue regarding this?

btw, my TFT module will only work without any shifting when i use 144 RED PCB and modify the offset from 32 to 0.

sumotoy, 2015-12-01:
This is a clear sign that cs pin sometime get floating, I have 2 boards here running from a couple of days with a wifi server on one and client on other, both with display, both with ESP8266-12E node mcu, still looks ok.
Did you use the D0 as cs pin? Maybe try to pullup d0 with a 10K resistor, I always do this since it's a good practice with SPI devices, it helps during startup when more than one device it's connected.

smetis, 2015-12-01:
Thanks for the quick response!
yes i am using D0 as CS pin. I just added a 10K pullup, but still see the distorted display and white screen after running for a while. I do feel like it is less frequent now. 

any other pins that need a pullup?

sumotoy, 2015-12-01:
Do you running something like wifi or something else concurrently? Or just the display alone?
Are you using the 2.0rc2 of Arduino ESP8266?

smetis, 2015-12-01:
no, i am running just the display alone now. i am not using rc2 either, using only the stable version. 
i added a cap to my power supply module and ran the clock example. 
now it seems to run for 30mins without white screen issue, but from time to time, the "Second" red hand will have residue on display. 

so i am still unsure if this is stable enough, will wait for another half a day and see if it is still running fine without white screen. The residue issue, still requires further investigation. any insight?

sumotoy, 2015-12-01:
Try to go in settings and enable _ESP8266_STANDARDMODE, maybe there's a problem with some module. I got only one user that has blocks but not white screens. But I can confirm it's stable here, already more than 24 hours 2 modules running.

smetis, 2015-12-01:
i actually read on that and already enable this mode. seems like not the same issue.

smetis, 2015-12-05:
Hi sumotoy,
just an update on this issue on github, it was the same question asked in esp8266 forum. 
For reference, this issue is suppressed after changing SPI speed to 8Mhz instead of 80Mhz.

sumotoy, 2015-12-05:
Hi, I got other 2 ESP8266 12E on nodemcu modules, had some issue when used with Arduino ESP8266 Option, probably firmware it's old.
I've updated to v1.3.0.2 AT Firmware.bin, then tested, works.
Now tested at maximum SPI speed, both working from yesterday nigh without stop. Actually I have 4 modules working at max SPI speed with no issues and till now you are the only one that need this setting so I don't think it's a library issue, I'm happy you solved the problem but must be something else.

closed 2015-12-05 by sumotoy

smetis, 2015-12-05:
what issue you see before update to 1.3.0.3 firmware? blank screen similar to mine? and what was the previous version of firmware that you see this issue?

sumotoy, 2015-12-05:
I don't tested with this library, just noticed that uploading create garbage on terminal, quite more than usual, so I immediately decided to change firmware.
Honestly I'm not familiar with ESP8266 firmware misteries, I was supposing that arduino IDE upload a new firmware together with sketch but documentation around this chip and forum don't really helped me a lot, so I normally upload with ESP_DOWNLOAD_TOOL_V2.4.exe  a terminal firmware that I trust (v1.3.0.2 AT Firmware.bin), reboot and use normally with Arduino.
It worked till now.