sumotoy/TFT_ILI9163C #19: CC3200 with RA8875 on energia

leedennis519, 2015-12-04:
Hey, 

I tried ur library with Arduino UNO and it works perfectly, however, I'm trying to use the same library for CC3200 on energia and I wanted to ask what are some of the stuff that I would need to change for the RA8875 library to be compatible with CC3200 ?

Cheers,
Dennis

sumotoy, 2015-12-05:
It's energia based if I'm not wrong?
I don't have CC3200, I have started time ago some kind of support for energia, I was suppose they provide a kind of SPI subset common to all devices but unfortunatly not so actually every device need his procedure, almost impossible without have phisically the device.
Actually it's not in my todo list, have to complete the 1.0 release, maybe in the future..

closed 2015-12-05 by sumotoy

leedennis519, 2015-12-05:
Cool, no worries I'll look into your stellaris framework, could probably
port it in there and do some minor hacks, thanks again.

sumotoy, 2015-12-05:
stellaris was never tested! It was compiling long time ago but never tested actually...