sumotoy/TFT_ILI9163C #24: Screen shifted down with Arduino Pro Mini

gimpox, 2016-01-15:
Hello,
The library won't work properly with my Arduino Pro Mini (original by Sparkfun, no clone).
The screen seems always shifted down of around 32 pixels.... (I've just run the test.ino sketch)

I have installed the latest library versions but nothing has changed, problem is still there:

Adafruit_GFX library = version 1.1.3
TFT_ILI9163C library = version 0.9
Arduino IDE = version 1.6.5

Wiring:

sck =pin 13
reset = pin 12
sda = pin 11
cs = pin 10
A0 = pin 9
GND = GND
VCC = VCC = 3.3V
LED = 3.3v

I've also tried to use several #define directives as indicated in your GitHub site but they seems useless.
I've also searched in the .h file for the magic "OFFSET" variable with no success.

[image]

I bought the display here in Germany (please ignore the wrong/ridicolous driver specification):
http://eckstein-shop.de/144-inch-SPI-TFT-LCD-Color-Screen-ST7735-Drive-IC128128

Here below the code of test.ino sketch I've used:
[test.zip](https://github.com/sumotoy/TFT_ILI9163C/files/92616/test.zip)

dobrMAN, 2016-01-16:
Chek settings in _settings/TFT_ILI9163C_settings.h and try change to #define __144_BLACK_PCB__ or #define __144_RED_PCB__ do not forget to comment out other #define.
Or use #define __144_RED_PCB__ and try change #define __OFFSET     32 in some file in or near the line 31.

gimpox, 2016-01-16:
SOLVED!

Hello dobrMAN, thank you for the hints. Yesterday night I've solved the problem, sorry for the delay in updating this issue.
My red tab is not a red tab but... a black tab! Even if painted red! 8|
So my TFT could be named as a "false red tab" display.

I've commented the line (in the file that you mentioned above)
#define 144_RED_PCB

and commented out the line
#define __144_BLACK_PCB__//128x128

So offset variable (as I understand) is now zero. It works like a charm, Great!

p.s: chinese manufacturers have a big fantasy :)

gimpox, 2016-01-16:
I'm going to close this issue. Problem solved.

closed 2016-01-16 by gimpox