sumotoy/TFT_ILI9163C #26: setRotation() doesn't work properly

gimpox, 2016-01-17:
Hello,

when setting the rotation to 0 (zero) all works fine on my Arduino Pro Mini 3.3V/8Mhz. But Today I had the idea to turn the screen upside-down by using setRotation(2) instrution. As result, the drawing area is rotated by 180° but the area is shifted down of 32 pixels also.

`setRotation(0)` output:
[image]

`setRotation(2)` output:
[image]

Here the simple test code:
```
void setup() {   
    tft.begin();

    tft.clearScreen();
    tft.setRotation(2);
    tft.fillTriangle(64, 0, 127, 127, 0, 127, RED); // triangle pointing upward
    delay(5000);
}
```

I've defined my ILI9163C 128x128 display by using 
`#define __144_BLACK_PCB__//128x128`
That is the only working choice. In fact, if I define the TFT as "red tab", the drawing area is shifted down of 32 pixel even when the rotation is set to zero.

My configuration:
IDE 1.6.5
Adafruit_GFX library version 1.1.3 (latest available)
TFT_ILI9163C-master library version 0.9

dobrMAN, 2016-01-18:
Uncomment Offset in lines 1076-1099:
```
void TFT_ILI9163C::setRotation(uint8_t m) {
	rotation = m % 4; // can't be higher than 3
	switch (rotation) {
	case 0:
		_Mactrl_Data = 0b00001000;
		_width  = _TFTWIDTH;
		_height = _TFTHEIGHT;//-__OFFSET;
		break;
	case 1:
		_Mactrl_Data = 0b01101000;
		_width  = _TFTHEIGHT;//-__OFFSET;
		_height = _TFTWIDTH;
		break;
	case 2:
		_Mactrl_Data = 0b11001000;
		_width  = _TFTWIDTH;
		_height = _TFTHEIGHT;//-__OFFSET;
		break;
	case 3:
		_Mactrl_Data = 0b10101000;
		_width  = _TFTWIDTH;
		_height = _TFTHEIGHT;//-__OFFSET;
		break;
	}
```
P.S.:
Using 1.0.5r better, i think.

gimpox, 2016-01-18:
Hello dobrMAN,
I have uncommented the lines above with no results as expected, since for the "black tab" __OFFSET variable is set to zero in the definitions section of TFT_ILI9163C_settings.h file . If I set it to 32 then the triangle shift down for further 32 pixels (64 pixels in total).
I've also tried other values (even negative!) with no success :(

dobrMAN, 2016-01-18:
Do u try define it as Red when uncomment offsets as i say?
1. Set settings as RED PCB (with offset 32).
2. Then uncomment it offsets in TFT_ILI9163C.cpp lines 1076-1099.

Any time, you can play with __OFFSET in lines 1076-1099 of TFT_ILI9163C.cpp, try replase it at +/-32
Note: any case in this piace off code is a rotation position
case 0: - default; case 1: -default+90degrees etc.

Sorry, it is only IMHO, i can't try this, because i have no free TFT.
And sorry for my englih, my favorit language is Russian, and i do not like use translators.

gimpox, 2016-01-18:
Uh! I forgot to set it as red tab. My fault.
Anyway no success, same result as before. The triangle can only shift further down, it never raise up (even with negative values).
I think my TFT is a pure 128x128 display.

Your english is ok to me!

gimpox, 2016-01-18:
My TFT it's really strange, looking at it with attention one can note that the rear side is different from the sumotoy's red tab (and from the majority of other red tab too). It has unique solder routes that are very different from the other ones.
Also, when issuing the readID command the 3-bytes length id returned is just 2 bytes long, i.e. "0x9101" (to obtain this I've got the help of another guy on the Arduino forum ). It seems not produced by Sitronix or other know manufacturers... :(

Here below an image of the "true" red tab (look at the solder trace):
[image]

Here my "false" red tab
[image]

sumotoy, 2016-01-31:
Looks like it's offset on top and bottom, it's the first strain of his species.
Can you give me the link where you buy it?

gimpox, 2016-01-31:
Here on ebay (they have also a website):

http://www.ebay.de/itm/281725377529?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

As you can see from photo above, what I received is NOT the one in the photo (even if the vendor accepted to have it back and refund me).

Further photos you can get from this thread on the arduino forum (skip to page 4):
https://forum.arduino.cc/index.php?topic=260605.45
(main differences: the color is not exactly red and the height is lower than the "true" red tab)

sumotoy, 2016-01-31:
Are you sure it's an ILI chip? They claim ST7735 on the product page!

gimpox, 2016-01-31:
The chip advised is wrong, download the datasheet from their website (as the vendor suggested to me):
http://eckstein-shop.de/144-inch-SPI-TFT-LCD-Color-Screen-128128_1
(Adafruit ST7735 libs doesn't work on it.)

sumotoy, 2016-02-07:
I've ordered and se what happen,

gimpox, 2016-02-07:
Ok. Thanks for your time (and money!)

sumotoy, 2016-03-09:
Well, the display arrived today but...surprise! It's the same I have here!

sumotoy, 2016-03-09:
Try look on the side of the display, there's numbers or may take a clear picture

gimpox, 2016-03-09:
Hy Sumotoy,
I'm surprised too, what it means "it's the same I have here"? They sent you exactly the one shown in the picture on ebay? They became smart! :) :)
Unfortunately my "false" red TFT is not here at home now. Tomorrow I will pick it and I will make some other photos. Anyway, as I remember, there was no numbers or codes on the side...

sumotoy, 2016-03-10:
The one I've got it's the same of the many I have already here and the same of ebay pictured, ergo, it works as always.

gimpox, 2016-03-11:
BTW, if you live in Europe I can send one of that TFTs to you by post.

sumotoy, 2016-03-11:
> BTW, if you live in Europe I can send one of that TFTs to you by post.
Maybe better, it takes too long from china and I have the feeling this has disappeared, it's really strange that display has no identification.
please send an email to sumotoy at gmail dot com

gimpox, 2016-03-11:
You've got an email.

dcpyatt, 2016-04-01:
I am experiencing the same issues with my red board from China.. same yellow headers, same traces...

Any resolve yet?  The 9340 graphicstest does address the whole screen, but text modes are mirrored.

Thanks!
DaveP in Alliance, Ohio, USA
dcpyatt@gmail.com

gimpox, 2016-04-01:
I've sent my strange TFT to Sumotoy by post one week ago. I think he is working on it now.

Anyway, I have orderd five TFT from China and I've got another type again! It is marked as "2.1" version and its dimensions are a little bigger.
Take a look here:
http://forum.arduino.cc/index.php?topic=260605.msg2690157#msg2690157

sumotoy, 2016-04-01:
Hello,
I still waiting it arrive but until 2 days ago nothing, I will call my mother tomorrow morning to see if there are news. It will arrive to my mother home and I will get the day after since I'm living in Venice and post it's not good here, I will tell you when I got it. It seems spreading this new version unfortunatly so I'm curious to see and solve.

gimpox, 2016-04-02:
Hi Max, I sent you an email.

dcpyatt, 2016-04-02:
OK, here is what I have came up with so far... perhaps it is just restating what you have already concluded... I'm sorry, I'm trying to get up to speed.

I have the same yellow headered module as gimpox pictured above.
I am running the TFT_ST7735-master library and so far the issues I see is the RED and BLUE are swapped.and scroll doesn't work, and when rotating the extra 32 is on the bottom.  Orientation for setRotation = 0 puts the top at the header pins.

Personally I might not need scroll, and I can adjust for the colors, but straight up text and graphics seem to work without bumping the extra Y+32.  I guess it would be nice to have it rotated 180 degrees so that pin 1 would be to the left.  Really cosmetic for my needs, but correctness is always preferred to a hack.

gimpox, 2016-04-02:
dcpyatt, check the little code fragment that I've posted on the Arduino forum to see the correct offset value for X/Y axis.

Question: how you restore the correct value for color? I have no idea at the moment.

dcpyatt, 2016-04-02:
Color is defined as '18-bit color'... kind of an odd concept when represented as a 16-bit number.

`RED =  1111 1000 0000 0000`
`GREEN = 0000 0111 1110 0000`
`BLUE = 0000 0000 0001 1111`

Notice that it is a 5-6-5 format because the eye can distinguish more in the green range than red or blue.

This actually works to our advantage as RED and BLUE are both represented by 5 bit numbers and can be swapped like for like.  If the color mix up included GREEN there would be a margin of error in converting from 5 bit to 6 bit and vice versa.

I've only had a couple hours of sleep, and may be going on an unplanned 3-1/2 hour each way road trip today, but I will work on a color swapping routine as time allows.

gimpox, 2016-04-02:
thank you dcpyatt,
let me know when/if you find an easy method to convert default library rgb565 colors automatically, without having to define everyone manually with the #define directive somewhere.

dcpyatt, 2016-04-02:
OK, here is the rough of it all, untested, but I believe logically sound:

`//COLOR is old incorrect color`
`correctRed = (COLOR && 0x0800) / 0x0800;`
`correctGreen = COLOR && 0x07E0;`
`correctBlue = (COLOR && 0x001F) * 0x0800;`
`correctColor = correctRed + correctGreen + correctBlue;`

incidently... the color that is correct, Green = DEC 2016! :)

gimpox, 2016-04-02:
Niceeeee! I don't know if it works but sounds so technical... it MUST work! :) :) :)
(I'm going to test it now)

dcpyatt, 2016-04-02:
Some errors in the boolean logic, but I am working them out...

dcpyatt, 2016-04-02:
OK, here it is in the form of a function and with a piece of demo code to go along with it.

Color Correction for the Yellow Headered Red TFT
By David C. Pyatt
Arduino Day 2016 - 4/2/2016

```
#include <SPI.h>
#include <TFT_ST7735.h>

// Color definitions
#define BLACK   0x0000
#define BLUE    0x001F
#define RED     0xF800
#define GREEN   0x07E0
#define CYAN    0x07FF
#define MAGENTA 0xF81F
#define YELLOW  0xFFE0  
#define WHITE   0xFFFF

#define __CS  10
#define __DC  9
#define __RST 8

TFT_ST7735 tft = TFT_ST7735(__CS, __DC,__RST);

void setup(void) {
  tft.begin();
}

void loop(){
  int color;
  
  //color is old incorrect color
  tft.clearScreen();
  color= RED;
  color = correctColorAdjust(color);
  draw_box(color);
  tft.setCursor(10,10);
  tft.print("RED");
  delay(1500);
  
  tft.clearScreen();
  color = GREEN;
  color = correctColorAdjust(color);
  draw_box(color);
  tft.setCursor(10,10);
  tft.print("GREEN");
  delay(1500);
  
  tft.clearScreen();
  color = BLUE;
  color = correctColorAdjust(color);
  draw_box(color);
    tft.setCursor(10,10);
  tft.print("BLUE");
  delay(1500);

  tft.clearScreen();
  color = YELLOW;
  color = correctColorAdjust(color);
  draw_box(color);
  tft.setCursor(10,10);
  tft.print("YELLOW");
  delay(1500);
  
  tft.clearScreen();
  color = CYAN;
  color = correctColorAdjust(color);
  draw_box(color);
  tft.setCursor(10,10);
  tft.print("CYAN");
  delay(1500);
  
  tft.clearScreen();
  color = MAGENTA;
  color = correctColorAdjust(color);
  draw_box(color);
  tft.setCursor(10,10);
  tft.print("MAGENTA");
  delay(1500);

  tft.clearScreen();
  color = WHITE;
  color = correctColorAdjust(color);
  draw_box(color);
  tft.setCursor(10,10);
  tft.print("WHITE");
  delay(1500);
}

int correctColorAdjust(int clr){

  int correctRed;
  int correctBlue;
  int correctGreen;
  int correctColor;
  
  correctRed = (clr & 0xF800)/0x0800;
  correctGreen = clr & 0x07E0;
  correctBlue = (clr & 0x001F) * 0x0800;

  correctColor = correctRed | correctGreen | correctBlue;
  
  return correctColor;
}
  
void draw_box(int color) {
  tft.drawLine(0, 0, 127, 127, color);   //TOPL BOTTOMR
  tft.drawLine(0, 127, 127, 0, color);
  tft.drawLine(0, 127, 0, 0, color);
  tft.drawLine(0, 63, 127, 63, color);    //MIDDLE
  tft.drawLine(0, 0, 0, 127, color);
  tft.drawLine(0, 0, 127, 0, color);
  tft.drawLine(0, 127, 127, 127, color);  //BOTTOM
  tft.drawLine(127, 0, 127, 127, color);
}

```

sumotoy, 2016-04-05:
Mmm, so the yellow PIN strain uses an ST7735 ?!?!
Are you using the mine ST7735 library?
I'm really freaking out to have one of this display in my hands, I have feeling that they put a new type of chip called XR6635 or similar that it's a low cost clone of the ILI one.

dcpyatt, 2016-04-05:
[image]
Here is the packaging...
Marked as QDM144SPINT7735RA

dcpyatt, 2016-04-05:
Found this on the Arduino forums... I wonder if he has tried rotation or scrolling, or even colors...

http://forum.arduino.cc/index.php?topic=378158.0

sumotoy, 2016-04-05:
I have a couple of ST7735 display here and they rotate correctly, also colors are OK.
I was living in a big city chinatown, I have learned don't trust what chinese write or say, eh eh...
Example:
1TB Hard Disk:
[image]

gimpox, 2016-04-05:
Incidentally, I have received other 3 TFTs few days ago that are identical to the ones received by dcpyatt (same package, same yellow pins).

sumotoy, 2016-04-12:
I have received the TFT today from gimpox (thanks man!). Tomorrow I will look into it.

sumotoy, 2016-04-12:
At first look, comparing with old one here's result:
1) Unidentified one it's smaller! the display and the whole PCB it's smaller than RED tab ones.
2) The flat connector it's different (size and pins are less).
3) They using a much cheaper 3v3 regulator. This regulator has high dropoff (about 1V) so it's not good for voltages less than 4.5V
Tomorrow I will try to understand something with code.

gimpox, 2016-04-12:
1. Yes I know, and I loved the reduced size.
2. I was not aware of that, thanks for the note.
3. Ouch! This means that powering the TFT with 3.3V it's a suicide, yes? There is a way to jump the voltage regulator and feed it with 3.3V?

sumotoy, 2016-04-12:
Yes, near regulator there's a small solder jumper called J1, short it and you can use 3V3.

dcpyatt, 2016-04-12:
I'm not sure what you are saying about the 3.3v... I have had no issues powering it from Arduino or Amica/NodeMCU 3.3v...  are you saying it is only getting ~2.3v?

sumotoy, 2016-04-12:
At 3V3 the dropoff can be 0.2V (since the regulator not working at that voltage it looks almost as short), this can be good but remember that logic signals cannot be more than VCC or chip can have some kind of damage if used massively (aka, fast SPI operations at 3V3 and display supplied at 2V8).
On the old RED PCB one there's a jumper onboard to bypass regulator and use 3V3, this board has the same jumper.
Btw I found these little display quite robust and I have noticed that ESD protection on pins act as bridge (when you don't have VCC!) and power up the display logic! This happened a couple of times, however it's dangerous so don't try at home!

dcpyatt, 2016-04-13:
So your recommendation would be to drop a little solder across to bridge those two contacts?  I want it as robust as possible as it will be a commercial endeavor.

sumotoy, 2016-04-13:
If you use 3V3 MCU and your MCU provide 3V3, better use the MCU supply since it's the same voltage level of the signals, I will go to short it.

gimpox, 2016-04-13:
I've noticed a strange thing on other type of TFT, i.e. the red-pcb, version 2.1, black-header: if I disconnect the wire powering the TFT at the Vcc pin then the display... continue to work without problems! How can it be???
It turn off only when I disconnect the power at the led pin too. 

I would to know if this strange behaviour happens with the "yellow-header" too.

sumotoy, 2016-04-13:
This can be an effect of the ESD protection of the input pins, there's mosfet (and probably some diodes) and this can happen if the logic level of signals is higher than VCC, signals are so busy that some current flow inside the mosfets and all together are able to sustain the logic supply demands, of course this should not happen. Maybe it's a good idea try to put a 100Ohm resistor in serie on each SPI line, have to try this.

gimpox, 2016-04-13:
I think that all the TFT pins recieve the same voltage at HIGH level. To test it, I've connected all the pins to an arduino mini pro 8Mhz/3.3V. To power the TFT and turn the led on I simply set the corresponding arduino pin to HIGH level. Even in this case the strange behaviour remains: disconnecting the wire at the Vcc pin the TFT continue to work. 8[

The version 2.1 is seems identical to the version 1.1 that you used (i.e. the "old" red pcb):

- same dimensions (i.e. slightly bigger than the "yellow header")
- black header
- red coloured pcb
- "version 2.1" is printed on rear of the PCB
- no serial number printed anywhere (this is a substantial difference with your tft)

At the moment no (german) vendors are able to sell you the "old" red pcb. I've made other two orders and they both sent me the 2.1 version, not the one in the photo.
The adafruit libs works for sure but I don't remember if I made a test with sumotoy libs too... :(

gimpox, 2016-04-13:
Just a note: voltage output at the arduino pin is never the same as 3.3V at the Vcc pin. Normally it is around 2.97-3.03 Volt (measured with the multimeter) 'cause of some dropout.

sumotoy, 2016-04-13:
This is a sign of bad design, they probably missed diodes inside ESD circuit in display circuit.
About the logic level of arduino, if supply it's 3v3 it should be very near 3v3, mus measure with an high impedance multimeter or the current flow will cause dropout.

gimpox, 2016-04-14:
Ooops! I forgot to mention that I measure 2.97/3.03 volt only when some load is attached at the pin (i.e. a LED dragging 10mA of current). When no load is connected to an output pin then the voltage is ok; it's pretty identical to the one applied at the Vcc pin.

That was just to say that the voltage at the logic pins driving the led is usually lower then the voltage at the Vcc pin. There is always some "load" (even minimal) dropping the voltage level just a little.

In any case this is a case of bad design as you wrote.

sumotoy, 2016-04-15:
Good news, after lot of hours I finally figure out to solve the yellow pin strain dilemma. It's an ILI9163C but display has 2 offset, this caused a lot of mumble thinking since the entire library it's not designed for double offset, so I have to redesign several part to include this strain (and possible other future ones). I will publish a new version of the entire library soon but first I need some beta testing so I will send to you an early version soon, it's ok to you?
Actually the version I'm testing now it's perfectly working on your display :8ball:

dcpyatt, 2016-04-15:
Any chance there will be any Lua API tips along the way?  I'm thinking I am going to be running on the Amica...

gimpox, 2016-04-15:
This are news nice to hear!
Are you planning to include it in the next "monolithic" 1.0 version?

sumotoy, 2016-04-15:
Sorry **dcpyatt**, never played with LUA, maybe in the future I will look at it.
**gimpox**, I will release in a day or two but tomorrow I'll send you a beta to test, if everithing as expected I will post the new version for all.
Upcoming version have several changes, support for almost any type of offset so it's virtually possible include any future strain, user settings has been expanded so it's far more easy to add yourself new type of display by creating another preset.

gimpox, 2016-04-16:
sumotoy, that will be the version that doesn't need adafruit libs at all? 

In any case your libs will be very useful because, as I can see, there is a lot of that yellow-pins TFTs around the world today.

sumotoy, 2016-04-16:
Of course yes, it's totally standalone. I have just finish the working beta, I need some more time to test with other MCU's I have here then I'll send to your email already setup for the yellow pin strain.
I have changed a lot of things in the user settings, now each display has it's own file in a separate folder called '_display'.
Offset management has updated, now can virtually manage every type of chinese weirdness since it's also rotation sensitive and code it's also simplified.
An user can create display settings for the upcoming strain by just duplicate one of the display settings file, change name, modify settings and include the file in the _settings/TFT_ILI9163C_settings.h
I hope he dosn't forget to send it here so I can include in any future release.

closed 2016-04-20 by sumotoy