sumotoy/TFT_ILI9163C #27: ESP8266 drawColorBitmap not working as expected

smetis, 2016-03-08:
Hi,
I created a bitmap array using lcd image converter and use drawColorBitmap to load it. The output image on screen is distorted. I can see the image but it is kind of skewed and shifted. Is it some timing issue? Appreciate if you can provide some clue in debugging this.

Note: I also found an error on the condition checking code in drawColorBitmap (pull request submitted), without fixing this the code will just return without drawing anything.

sumotoy, 2016-03-08:
No, this happen because for some reason the width of the array you created probably it's not exact 48 and results in a skewed image. 
Try think about an old TV, the cannon has to move from top right and when arrive to with of the image it has to go back and advance of one row and so on, this is what happen with the function.
Try set 47 or 49 as width

smetis, 2016-03-08:
I am pretty sure the array is 48x48. I even created a 10x10 array that I can easily count and verified, the same issue occurred. It get skewed. Have you get a chance to try drawcolorbitmap using esp8266?

sumotoy, 2016-03-08:
did you try to put 47 or 49 as width?

smetis, 2016-03-08:
nope, not having the setup with me now. will give it a try this evening and report back.

sumotoy, 2016-03-08:
If works, please post here the array in question and I will investigate

smetis, 2016-03-09:
yes, it works when i changed to 47. 
And I think I know why. According to the datasheet on setting the display window, the X start and X end is the index of the window. So I believe the code in drawColorBitmap should offset the width and height when setting the window.

Below is my suggested changes:
`setAddrWindow_cont(x,y,w + x - 1,h + y -1);//constrain window`

I don't understand how it works previously, but it looks like setting w+x and h+y will definitely create a shift of one pixel.

Let me know your thoughts. Thanks!

sumotoy, 2016-03-09:
Well, I already know this but the function will be replaced soon with icon support for the next release. The function was intended as quick & fast example and survived trough releases because I've loosed time on font rendering optimizations.

closed 2016-03-09 by sumotoy

smetis, 2016-03-09:
great, thanks for the confirmation. Will you make changes to the current code or you plan to just release the replacement function along with icon support?

sumotoy, 2016-03-09:
The icon support will replace this, I plan an _icon folder and user can just load icons inside sketch.
It's also easier since the conversion into icons will be easier.

smetis, 2016-03-09:
cool, this is what I am using the library for now, to display icon. I assume the icon files still need to be in 24bits RGB array form, right?