sumotoy/TFT_ILI9163C #28: TFT powering problem

dmitry335, 2016-03-09:
Here is a problem I got with ILI9163 with black pc on arduino nano.

I connected TFT through level-shifters all lines. VCC and LED directly to 3.3v

I could make it work only by connecting 3.7 battery directly to +5V or Vin pin of arduino.

Powering from USB - gives white screen.
Powering from 5.0 V power suply to +5V pin - gives white screen
Powering from 5.0 V power suply to Vin pin - gives flickering screen with wrong rotation.

here is video:
https://youtu.be/QYVGRAo7i6c

what is wrong?
thank you

sumotoy, 2016-03-09:
Many chinese 128x128 has a 5V->3v3 regulator on the back, this mean you should power display at 5V.
My RED one and another BACK one I have here have both regulator on the back and really near regulator there's a minuscole solder jumper, if it's open (and you have the tiny regulator soldered) you should supply at 5V, if it's closed you should provide 3V3.
Supply 3V3 on a TFT with a 5V->3v3 regulator in most case result in insufficent supply for the display logic, the backlight will light just fine but logic not.
Check if you have the regulator on the back.

dmitry335, 2016-03-09:
Yes, I have regulator. Will it be OK to drive all logic lines at 5v? Or I
will fry TFT? There are no regulators on CS, SDA... pins.

Thanx

sumotoy, 2016-03-09:
NO, logic lines remains at 3V3!

dmitry335, 2016-03-10:
I have tried connecting TFT power to 5+ and it works the same.

And it works OK if powered to 3.7v through regulator and logic through
shifter , so that there are less than 3v

sumotoy, 2016-03-10:
dmitry, can you make a schematics of the whole setup? Included pullups, etc.

dmitry335, 2016-03-10:
here is schematic

[image]

sumotoy, 2016-03-10:
Oh Oh....
1) Why you are using pulldown resistors? You should take away all resistors!
2) If you are using other SPI devices you need only a pullup resistor (+3v3 to CS) Otherwise useless.
3) The level converter you are using are not good for SPI, this are designed for bidirectional I2C (max 400K), the fet will almost destroy the SPI waveform. Better use a simple CD4050 powered at 3V3 (it's 5V in tolerant).
4) You can get rid of the Reset Line, just connect a resistor (4K7 to 10K, any value) between RST to +3V3

dmitry335, 2016-03-10:
Wow!! Thank you a lot!!! I'm new with this stuff, and sometimes I look answers for simple questions very long.
:)

sumotoy, 2016-03-10:
In digital electronics mostly pullups resistors are used, very rarely pulldown (from signal to ground), this because many digital stuff are 'tristate' that mean 'high impedance' (I mean very high, can be 10Mohm). This cause the digital input very unstable, so a 'pullup' resistor will stabilize the line. 
For avoid interferences most digital stuff use the 'Hi' state as 'disabled', see the many switch examples around, they became active when shorted to ground.
On I2C it's popular use pullup resistors (between 2k2 to 5.1K) on both SDA and SCL, I2C it's relatively high capacitive and needs lower resistors in relation to speed but SPI lines are mostly very high speed and only CS needs really a pullup (remember? for CS the Hi State means 'disabled)', this is really useful only when you sharing the same SPI lines with several device since different devices commands will not interfere between each other.

closed 2016-04-16 by sumotoy