sumotoy/TFT_ILI9163C #29: 1.0r6?

larsenglund, 2016-04-18:
It's monday now... ;)
Thanks for your awesome work with this library!

sumotoy, 2016-04-18:
I'm currently tuning color parameters, each display's have different settings, so I definetively release today.
It's a big update, structure changed a lot, all SPI calls are unified and separate between processors and now it's far more easy adapt different (future) display's since each one has it's own file.

larsenglund, 2016-04-18:
Awesome, looking forward to trying it out!

sumotoy, 2016-04-18:
1.0r6 changes:

- Compatibility with RED PCB Yellow pin strain
- Changed overall library structure, easier to add CPU's
- Separate display's files, now each display has full set of parameters, easy to add more incoming strain
- Handle 18Bit color depth
- Changed a bit Font Scheme, new ones are backward compatible, old ones should to be re-converted with lcd-image-converter (last beta) and with new template (sumotoy_font2.1.tmpl). I have also added a conversion preset since the last beta of lcd-image-converter has this feature (Font_Conversion.xml).
However I have included an option in user settings that force lib to read old font types but re-conversion should be performed as soon as possible since the release will not support this.
- Included font files for editing, if you find any problem you can edit directly library font files with lcd.image.converter and compile for library.
- Library use less space, made a lot of optimizations on code
- Small error corrected
- Speed almost unchanged, something faster, something slight slower but almost same.
- Delete unused and useless functions

closed 2016-04-18 by sumotoy