sumotoy/TFT_ILI9163C #33: setCursor orientation-dependent ? (1.0r6)

reaper7, 2016-04-26:
when I use setCursor(x, y) then I expect that first value is x axis and second value is y axis (no matter how is set orientation)
but now it is dependent on the screen orientation and I must flip x and y on setCursor when I change orientation.

for me setCursor and maybe other functions must check what orientation is set and (in this case) flip x and y inside function without user intervention.

check this example (I past loop only):
in this example I only change "y" value for setCursor for each of four orientation,
orientation 0 and 2 works as expected but 1 and 3 show wrong
```
void loop(void) {

//ROTATION 0
  tft.setRotation ( 0 ) ;
  
  tft.setCursor ( 0, 0 ) ;
  tft.print ( "line 1" ) ;
  delay(1000);
  tft.setCursor ( 0, 30 ) ;
  tft.print ( "line 2" ) ;
  delay(1000);
  tft.setCursor ( 0, 60 ) ;
  tft.print ( "line 3" ) ;
  delay(1000);

  tft.fillRect ( 0, 0, 128, 128, BLACK ) ;

//ROTATION 1
  tft.setRotation ( 1 ) ;
  
  tft.setCursor ( 0, 0 ) ;
  tft.print ( "line 1" ) ;
  delay(1000);
  tft.setCursor ( 0, 30 ) ;
  tft.print ( "line 2" ) ;
  delay(1000);
  tft.setCursor ( 0, 60 ) ;
  tft.print ( "line 3" ) ;
  delay(1000);

  tft.fillRect ( 0, 0, 128, 128, BLACK ) ; 

//ROTATION 2
  tft.setRotation ( 2 ) ;
  
  tft.setCursor ( 0, 0 ) ;
  tft.print ( "line 1" ) ;
  delay(1000);
  tft.setCursor ( 0, 30 ) ;
  tft.print ( "line 2" ) ;
  delay(1000);
  tft.setCursor ( 0, 60 ) ;
  tft.print ( "line 3" ) ;
  delay(1000);

  tft.fillRect ( 0, 0, 128, 128, BLACK ) ; 

//ROTATION 3
  tft.setRotation ( 3 ) ;
  
  tft.setCursor ( 0, 0 ) ;
  tft.print ( "line 1" ) ;
  delay(1000);
  tft.setCursor ( 0, 30 ) ;
  tft.print ( "line 2" ) ;
  delay(1000);
  tft.setCursor ( 0, 60 ) ;
  tft.print ( "line 3" ) ;
  delay(1000);

  tft.fillRect ( 0, 0, 128, 128, BLACK ) ; 
}```

sumotoy, 2016-04-26:
Yes, thanks for point me this!
However I will release a new version this week based on 1.0p7 of TFT_ILI9163C that have this fixed plus much more. Code it's almost same, just initialization a some command.

reaper7, 2016-04-27:
tnx for info and great lib!
btw. please put this fix to Your TFT_ST7735 lib too ;)

sumotoy, 2016-05-03:
Yes. Plan is finish and test 1.0p7 then I can easily transfer to many libs!

sumotoy, 2016-05-03:
In the meantime you can put the new setCursor function:
`
void TFT_ILI9163C::setCursor(int16_t x, int16_t y) 
{
	if (boundaryCheck(x,y)) return;
	if (_portrait) swap(x,y);
	setArea(0x0000,0x0000,x,y);
	_cursorX = x;
	_cursorY = y;
}
`

reaper7, 2016-05-03:
ok, tnx!

closed 2016-05-25 by sumotoy