sumotoy/TFT_ILI9163C #35: ESP8266 pin name confusing

beegee-tokyo, 2016-05-05:
I am struggling with the connection of the display to my ESP8266 (Adafruit HUZZAH ESP8266 => https://www.adafruit.com/products/2471).

In your readme.md (Pre-Release-1.0r6) you write:

> TFT side   -------------------- ESP8266  
> - Vcc       -->     +3V3V(!!!!)
> - Gnd       -->     Gnd
> - CS        -->     D0
> - RST       -->     D2
> - A0        -->     D1
> - SDA       -->     Mosi (D7)
> - SCK       -->     Sclk (D5)
> - LED       -->     Some display need a resistor (see note below)

What is D0, D1, ... refering to?
It cannot be the GPIO numbers, because on the ESP8266 MOSI is on GPIO13, while you write D7.

Can you clarify please? Thank you.

sumotoy, 2016-05-05:
ESP8266 has no standard, many vendors create a board and put pinouts as they like to.
For ESP8266 development I use well documented official boards, like **NodeMCU 0.9**
[image]
This one uses ESP8266-12
Or **NodeMCU 1.0**
[image]
(uses ESP8266-12E)
For this library I'm using Hardware SPI (HMOSI,HSCLK).
You can save a pin, the RST on display side can be tied up by connecting a resistor (4K7 to 10K) to 3V3.
You still need 2 pin for CS and RS(A0), any unshared pin will work (for unshared I mean, pins that doesn't have any double function or pullup/pulldown).
Adafruit has it's way (notice that GPIO4/5 are inverted in schematics and rev A boards)
Here's the decoded pin map for Adafruit (btw not tested) accordly schematics:
SDA->GPIO13(MOSI)
SCL->GPIO14(SCLK)
For next, use any unshared,unpulled(up/down) pin
A0->GPIO4
CS->GPIO16 (I know, this is shared, but on NodeMCU works...) or GPIO5

There's tons of ESP8266 boards out there, everyone use his pinout scheme, Adafruit ones are the most expensive and because of this not as popular as the nodeMCU.

closed 2016-05-05 by sumotoy

beegee-tokyo, 2016-05-05:
Hi Sumotoy,
thanks for this info, I will try to connect the display with your information.
In my country (Philippines) the NodeMCU is the most expensive ESP8266 board and the Adafruit Huzzah is easier and cheaper to get.
But now I understand your definitions of D0, D1, ... and can adapt to the Adafruit board.

sumotoy, 2016-05-05:
I would like but almost impossible. Many vendors don't even publish schematics!
In an year of so, thanks to Igrr and his frontend, it will be possible have some kind of standard-like.
Even with schematics, infos are not complete. Inside ESP8266 there's a SPI Flash memory sharing SPI that can be affected by devices using SPI Mode 3, some vendors has 1M, some other 4 or more....
How much you pay Adafruit in your country?

beegee-tokyo, 2016-05-05:
At least in Arduoino IDE the pin numbers are corresponding to the GPIO
numbers.

Here the NodeMCU is around 18 Euros, the Adafruit Huzzah is around 16
Euros. Both from local distributors. Buying online (e.g. eBay) is more
expensive here because of the corrupt local customs that add random
additional fees to imports.

Well, that is one of the issues if you live in a third world country.
Prices are going up as soon as the sellers/officials see that I am a
foreigner (German). Cannot hide my nationality.

Mit freundlichen Grüßen
regards

Bernd Giesecke ﾍﾞァﾝﾄ  ｷﾞｾｹ
Sent by my Android device