sumotoy/TFT_ILI9163C #42: Can't get it to work :(

tymek22, 2016-06-11:
Hi,

I have the black pcb V1.1 display, but I can't get it to work. I use an atmega328p with internal 8MHz oscillator and first flash the compiled example. Then I make the connections and power it with 3.3V.
I tried both reset pin to 3v3 and via 10k resistor and both didn't work. I just get a white screen and noting else. Can someone help me please? How to make it work?
(I also ordered a new display because I thought that this one may be defect, but now I see that it's V2.1 so for now I can't use it.)

sumotoy, 2016-06-11:
never see the black v1.1!
The red pcb 2.1 it's an ST7735 and I have a library for it https://github.com/sumotoy/TFT_ST7735/tree/1.0p1
I cannot help without a link and more infos about this display.

tymek22, 2016-06-11:
Oh ok, thanks for that info, I didn't know that there are different versions of the black pcb. I will try the new library with the red one and will post the link if I can find it.

sumotoy, 2016-06-15:
Try use the ST7735 as well, recently chinese uses anything and PCB color and writings on back have no more sense.
https://github.com/sumotoy/TFT_ST7735

closed 2016-06-15 by sumotoy