sumotoy/TFT_ILI9163C #45: Rectangle on top of screen

arturgoms, 2016-07-09:
In my teensy 3.2 and 1.0.7 I'm getting a rectangle on top is screen.. I already set the settings.h but no success.

arturgoms, 2016-07-09:
[image]

sumotoy, 2016-07-09:
It's a RED PCB but yellow pin or black pin one?
Yellow Pin: decomment #include "../_display/TFT_ILI9163C_RED_PCB_YPIN.h"
Balc pin: decomment //#include "../_display/TFT_ILI9163C_RED_PCB_OLD.h"
And it works, I have both TFT here and the tft in the picture is a clear sign that has the wrong offset (32 pixels, I can beat on it)

arturgoms, 2016-07-09:
I try with all includes, I don't know what I'm doing wrong, I have de red one with black pins.. V1.1

Led 3v3 
Sck 13 
Sda 11 
A0 20 
Reset 3v3 
Cs 21 
Gnd gnd 
Vcc 3v3 

Any idea? 

Thank you.

sumotoy, 2016-07-09:
Did you noticed any change on display when change settings? I mean, the rectangle change position?
It SHOULD, because you are changing the offset of the whole screen. Please report this.
If this doesn't happen, maybe you have 2 version of the library, one inside Arduino/hardware/teensy/avr/libraries/ and another one inside your sketches folder/libraries/

An user reported a RED PCB black pin v1.1 but was an ST7735, I believe that all the latest version use chinese unbranded chip clones so they put whatever they have on hands inside, maybe you can try my ST7735 as showed here:
https://github.com/sumotoy/TFT_ILI9163C/wiki
It's exact the same library but for ST, please be sure to use the 1.0p
It's too early now to say "...uh, oh, here's another display strain..." but in the last year chinese changes this display so many times that forced me to introduce an universal workaround for present and future offset.

arturgoms, 2016-07-09:
YOU ARE RIGTH, There was two libraries, i just uncomment the library in avr folder and booom, WORKS!! Thank you so much.

sumotoy, 2016-07-09:
cool !

closed 2016-07-09 by sumotoy