sumotoy/TFT_ILI9163C #57: Display out of position

revolutionxx93, 2017-07-18:
Hey there,
I've got a problem with all example programs.
The programs are out of position, not in the middle. 
And the rest of the screen is shown weird pixels.
Is is posible to adjust the offset of the X or y position?
Does anybody got an Idea?

blackw212, 2017-07-19:
Check the user_setup.h and be sure to pick the correct model of display you have. I encountered the same problem, at first i thought my display is an ili9163, as the seller wrote on the website where i bought it from, but it turned out to be another one. I think you should try more options till you find one who's fitting to your display. I'm on a mobile phone now, therefore I can't be more precise than this, but I'll take a look and reply asap with the things which you should watch out for !

blackw212, 2017-07-19:
So damn, sorry for my reply, i thought it is a different library.
I've had a lot of problems using this library, with the offsets and other shit things like that. There's a new and better library for this, called TFT_eSPI made by Bodmer, and there is the thing i was describing earlier. It's much more better made than this one, so I'd suggest giving a shot at that one. This library is a bit outdated and i don't think OP is actually making any changes to it.

Try https://github.com/Bodmer/TFT_eSPI

revolutionxx93, 2017-07-19:
Thank you very much!
It was a wrong library.
It was a ST7735 not a 9163C.
But I also try the eSPI library you send me :)

blackw212, 2017-07-19:
Same thing happened to me, producer said it's an Ili9163 but in fact it is ST7735 - beware of the label color on it, it's really important for the offsets , they are different for each and ever color !