sumotoy/TFT_ILI9163C #66: 128x128 red board offset confusion

ecolyx, 2018-05-10:
My display didn't work with any of the defined board types/resolutions, so I hacked your code and got it working in each rotation. I split OFFSET into OFFSETX and OFFSETY, as the offset is different in landscape and portrait orientations, and then wrote an address pair for each rotation. The code works, but doesn't make sense to me, and I can't port it to the other boards, so haven't submitted a pull request. This will break the code when other boards are defined. 

I would be happy to modify and submit if someone wants to explain how the g.ram is being mapped to the display in this situation, and whether the code can actually be written in a more meaningful manner, and make it compatible with the other pcbs. 

I expected to end up with maybe two code section, for horizontal and vertical orientations, but the offsets requried in each rotation that work don't seem to allow for this to happen..

in the .h file
`
#if defined(__144_RED_PCB__)
...
	#define __OFFSETX		2//*see note 2
	#define __OFFSETY		1//*see note 2
`

in the .cpp file
`
void TFT_ILI9163C::setAddrWindow(uint16_t x0, uint16_t y0, uint16_t x1, uint16_t y1) {
	#if defined(__MK20DX128__) || defined(__MK20DX256__)
		SPI.beginTransaction(ILI9163C_SPI);
		_setAddrWindow(x0,y0,x1,y1);
		SPI.endTransaction();
	#else
		writecommand(CMD_CLMADRS); // Column
		if (rotation == 0){
			writedata16(x0 + __OFFSETX);
			writedata16(x1);
		} else if (rotation == 1) {
			writedata16(x0 + __OFFSETY);
			writedata16(x1 + __OFFSETY);
		} else if (rotation == 2) {
			writedata16(x0 + __OFFSETX);
			writedata16(x1 + __OFFSETX);
		} else {
			writedata16(x0 + 3);
			writedata16(x1 + 3);
		}

		writecommand(CMD_PGEADRS); // Page
		if (rotation == 0) {
			writedata16(y0 + __OFFSETY);
			writedata16(y1 + __OFFSETY);
		} else if (rotation == 1) {
			writedata16(y0 + __OFFSETX);
			writedata16(y1 + __OFFSETX);
		} else if (rotation == 2) {
			writedata16(y0 + 3);
			writedata16(y1 + 3);
		} else {
			writedata16(y0 + __OFFSETX);
			writedata16(y1 + __OFFSETX);
		}
		writecommand(CMD_RAMWR); //Into RAM
	#endif
}
`

SurenBono, 2019-06-27:
[image]

SurenBono, 2019-06-27:
[image]
[image]

vsbogd, 2019-06-28:
It seems, that I have the same issue as @ecolyx, at least I ended up with the same fix as he did.

@SurenBono, making `__OFFSET == 0` solves issue with large noised bar at the top of the display as your pictures shows, but after that two artifacts are still there:
- image shifted 2 pixels to the left, and 2 pixels noised vertical lines on the right;
- noised line ~1 pixel width at the bottom;

SurenBono, 2019-06-29:
Than check your width and height dimensions is it 128 x 160..?..or is still
defined as 128x 128... this SPI TFT ili9163c is working on wemos d1r1 ..and
should be manufactured as a stackable version.. i purchased a stacked
version last time but it was 8 bit..incompatible with esp8266  and esp32
...only ran on uno

>
> It seems, that I have the same issue as @ecolyx, at least I ended up with
the same fix as he did.
>
> @SurenBono, making __OFFSET == 0 solves issue with large noised bar at
the top of the display as your pictures shows, but after that two artifacts
are still there:
>
> image shifted 2 pixels to the left, and 2 pixels noised vertical lines on
the right;

vsbogd, 2019-06-30:
> Than check your width and height dimensions is it 128 x 160..?..or is still defined as 128x 128...

Yes, I have checked it, my config is:
```c
	#define _TFTWIDTH  		128//the REAL W resolution of the TFT
	#define _TFTHEIGHT 		128//the REAL H resolution of the TFT
	#define _GRAMWIDTH      128
	#define _GRAMHEIGH      160
	#define _GRAMSIZE		_GRAMWIDTH * _GRAMHEIGH//*see note 1
	#define __COLORSPC		1// 1:GBR - 0:RGB
	#define __GAMMASET1		//uncomment for another gamma
	#define __OFFSET		0
```

On the photo below it can be seen that:
1. Both top and left yellow lines are not visible
2. Noise points at the right side of the screen

(Sorry for a photo quality, could not get anything better)

[image]

SurenBono, 2019-07-01:
Try a different sketch...the Clock and see if it still persist...or
uncomment the black definition instead of red in _setting

           © SROTO&GARGEES™ <https://srotogargees.business.site/> c/o Suren

ecolyx, 2019-07-01:
I assure you I tried all these suggestions before modifying the code with seperate offsets.

Perhaps you could look at what my code changes and why it appears to work for us, when none of the suggested settings do work?

SurenBono, 2019-07-04:
Return and refund..mine worked just fine.initialy it had an offset
error.later fixed it ..i think the manufacturer messed up on its labelling
and everything else such as your issue...cant they make one that is
compatible for all board..uno,esp8266 & esp32 uno styled stacked SPI

vsbogd, 2019-07-04:
BTW, @SurenBono , which version of TFT_ILI9163C code do you use? Do you use `master` branch or another one or tag?

SurenBono, 2019-07-06:
Sumotoys version....but i purchased it for my version of edzelf esp radio
on D1R1 which originally writen for nodemcu..check it out..

SurenBono, 2019-07-06:
This was the wiring diagram that worked..

SurenBono, 2019-07-06:
Finally
..try to define outside the box 130 x 165 mybe its more than the default
128x160 dimension for your display...or count the pixel to check

SurenBono, 2019-07-06:
Or redefine it smaller bypassing the noise spectrum.. example= 127x159
...etc

vsbogd, 2019-07-08:
This configuration makes "fill by rectangles" test looks perfect, but on text test it can be seen that image is still shifted to the right:
```c
	#define _TFTWIDTH  		131//the REAL W resolution of the TFT
	#define _TFTHEIGHT 		128//the REAL H resolution of the TFT
	#define _GRAMWIDTH      131
	#define _GRAMHEIGH      160
	#define _GRAMSIZE		_GRAMWIDTH * _GRAMHEIGH//*see note 1
	#define __COLORSPC		1// 1:GBR - 0:RGB
	#define __GAMMASET1		//uncomment for another gamma
	#define __OFFSET		1
```
To be clear this text test looks equally shifted with config above and with config from https://github.com/sumotoy/TFT_ILI9163C/issues/66#issuecomment-507057733

[image]

SurenBono, 2019-07-08:
Ok..great...now start your cursor above zero..if normally set cursor (0,0)
then start with example ( 5,5)...or so

>
> This configuration makes "fill by rectangles" test looks perfect, but on
text test it can be seen that image is still shifted to the right:
> #define _TFTWIDTH  131//the REAL W resolution of the TFT
> #define _TFTHEIGHT 128//the REAL H resolution of the TFT
> #define _GRAMWIDTH      131
> #define _GRAMHEIGH      160
> #define _GRAMSIZE _GRAMWIDTH [...]
with config from #66 (comment)

SurenBono, 2019-07-08:
...those fillbyrectangles are codes for clearscreen for landscape mode
which is (3) and portrait as (1)....why not code it in the source library
as such that we only need to mention tft.clearscreen for both ..

ecolyx, 2019-07-09:
"I would be happy to modify and submit if someone wants to explain how the g.ram is being mapped to the display in this situation"

Would be much easier if someone could respond to this. Then we would not need to reverse engineer by trial and error.

Having worked for ARM Holdings as a core code maintainer, I suggest this should be much easier to solve than following the current methodology

vsbogd, 2019-07-09:
> Ok..great...now start your cursor above zero..if normally set cursor (0,0) then start with example ( 5,5)...or so
What is the goal of the test? I am 99% sure that I will see normal picture when will start from (2, 1) position.

vsbogd, 2019-08-04:
Finally, I have found that my TFT has `ST7735` controller not `ILI9163C`. @ecolyx, probably you also can try https://github.com/adafruit/Adafruit-ST7735-Library instead. 

Sorry for confusion, not an issue for me anymore.

ecolyx, 2019-08-04:
Oh, thankyou.

Interesting discovery, I will investigate when I get a chance to review this project. 

Is there a simple way to identify the controller?

vsbogd, 2019-08-05:
> Is there a simple way to identify the controller?

Yeah, nice question :-) I see that sometimes it can be deduced by reading device registers, like in https://github.com/prenticedavid/MCUFRIEND_kbv/blob/master/examples/LCD_ID_readreg/LCD_ID_readreg.ino But not sure about general case.

ecolyx, 2019-08-05:
Thankyou, I'll investigate :-)

SurenBono, 2019-08-06:
Mine is not stm display but the original ili9163c module and its working
perfectly .thnks for your concern

punund, 2020-06-08:
> Finally, I have found that my TFT has `ST7735` controller not `ILI9163C`. @ecolyx, probably you also can try https://github.com/adafruit/Adafruit-ST7735-Library instead.
> 
> Sorry for confusion, not an issue for me anymore.

Are you sure it's ST7735?  I have the same display and the same issue, but it won't work with ST7735 libraries.

ceptimus, 2022-07-25:
I have a red board with yellow headers. Works perfectly (whole display filled with no noise on any edges) with Adafruit_ST7735 library.  Define the right pins for your connection.  I'm using a (3.3V 8MHz) Pro Mini with:

#define TFT_CS 10
#define TFT_RST 8
#define TFT_DC 9
and
tft.initR(INITR_144GREENTAB);

Text on back says of board says, 1.44' SPI 128*128 V1.1

eletroeduardo, 2022-11-21:
Hi!

Just for record, having the same issue as @vsbogd, my solution was:

`__OFFSET 32 -> 0` (same as previously stated)
in TFT_ILI9163C.cpp's `setAddrWindow` function, bettwen` #else` and `writecommand`, add
```
x0 += 2;
x1 += 2;
```

jossueIvan, 2024-02-13:
se puede conectar a un esp32 c3 super mini y si si como?