sumotoy/TFT_ILI9163C #70: Issue with other library

Jim-bee, 2019-07-15:
Hi,
Your library works great with EastRising 1.8" display a couple of exceptions;
1) the example scroll function does not work for me.
2) Compatibility issue with MLX90393 library //From https://github.com/tedyapo/arduino-MLX90393 by Theodore Yapo.

NOTE: Both libraries work fine on their own.  Please let me know if this can be fixed or maybe a work around.

Compile Errors:

n file included from /Users/JimD/Documents/Arduino/libraries/TFT_ILI9163C-master/TFT_ILI9163C.h:135:0,
                 from /Users/JimD/Documents/Arduino/RewinderTS/RewinderTS.ino:11:
/Users/JimD/Documents/Arduino/libraries/TFT_ILI9163C-master/_settings/TFT_ILI9163C_registers.h:6:22: error: expected identifier before numeric constant
 #define CMD_NOP      0x00//Non operation
                      ^
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:38:5: note: in expansion of macro 'CMD_NOP'
     CMD_NOP = 0x00,
     ^
/Users/JimD/Documents/Arduino/libraries/TFT_ILI9163C-master/_settings/TFT_ILI9163C_registers.h:6:22: error: expected '}' before numeric constant
 #define CMD_NOP      0x00//Non operation
                      ^
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:38:5: note: in expansion of macro 'CMD_NOP'
     CMD_NOP = 0x00,
     ^
/Users/JimD/Documents/Arduino/libraries/TFT_ILI9163C-master/_settings/TFT_ILI9163C_registers.h:6:22: error: expected unqualified-id before numeric constant
 #define CMD_NOP      0x00//Non operation
                      ^
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:38:5: note: in expansion of macro 'CMD_NOP'
     CMD_NOP = 0x00,
     ^
In file included from /Users/JimD/Documents/Arduino/RewinderTS/RewinderTS.ino:14:0:
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:65:12: error: expected unqualified-id before ')' token
   MLX90393();
            ^
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:116:1: error: expected unqualified-id before 'private'
 private:
 ^
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:138:3: error: expected unqualified-id before 'private'
   private:
   ^
/Users/JimD/Documents/Arduino/libraries/arduino-MLX90393-master/MLX90393.h:141:1: error: expected declaration before '}' token
 };
 ^
exit status 1
Error compiling for board Adafruit Feather M0.

SurenBono, 2019-07-15:
...those are compilation error...you have to debug all of those error
before you can get what you want....belive me i seen a 1000 of those
myself...finding out how to fix em...will take some time...and it was not
my sketch either...im one of you..

Jim-bee, 2019-07-17:
Hello,

I have received information from Ted Yapo MLX90393 library guy.  He mentioned the following about your ILI90393 library in his reply to this issue.

"The other library should not pollute the global namespace with #defines like CMD_NOP. If they used scoped enums like this one does, this kind of error would not happen. “

I have proven this out that if I comment out your //#define CMD_NOP = 0x00 in your .h file then everything complies fine.  Will this cause any issues with your library or is this a good work around?

Please consider his recommendation to make your library more compatible with the other libraries out there.

Best Regards,
Jim

SurenBono, 2019-07-19:
I did not write any drivers or know how to for ILI90393.... must be a
mis-info...i was there regarding offset noise err on ILI9163C