sumotoy/TFT_ILI9163C #72: Screen width not fully utilized.

clparrish, 2019-12-13:
I built the Edzelf/ESP-radio using an ST7735 1.8" TFT display. The screen width is not fully utilized, however. For example, the top line looks like this:

                                                 ** Internet Radio
                                                 **
As you can see, the trailing asterisks are on the second line, indicating that the display thinks there is no room for them on the first line, which is not the case. Additional text on the display does the same thing. I'm using the TFT_ILI9163C library, as specified in the sketch.
Has anyone encountered and this same problem?

oSkrobuk, 2020-07-18:
Edit the settings in the TFT_ILI9163C_settings.h file for your display. Set 0 for offset

clparrish, 2020-07-20:
Thank you for the info. I'll try your suggestion as soon as i receive a new
ESP8266.
Curt