sumotoy/TFT_ILI9163C #75: BigPiture issue displays from half center of screen

Prathamesh08glovatrix, 2022-12-05:
I have tried all the library present for the TFT_ili9163C but BigPiture example itself giving issue it display image from half screen and again came to front. 
I am using ESP32 DEV MODULE. 
pins used 
#define __CS 5
#define __DC 2
rest connected to 3.3 v
BigPiture example itself not working properly.
can please anyone help me on this.
thanks.

https://user-images.githubusercontent.com/115615283/205622655-f209467b-e1f6-4ba3-9d33-888c50006722.MP4

[image]

Abdullah-python, 2023-08-03:
I am facing the same issue, I tried to use the SD card example with a few tweaks but it still wont display properly. Let me know if you found a solution yet.

This is my progress so far with the SD card example
[image]

Abdullah-python, 2023-08-04:
Try the Adafruit Imager library. It worked for me. I am attaching the link below to the library tutorial. The only drawback is that you will have to use a SD card. Im not sure if you have a limitation in adding the SD card.

Hope this helps

https://learn.adafruit.com/adafruit-1-44-color-tft-with-micro-sd-socket/drawing-bitmaps